itu-auv/auv-software #87: investigate/fix gazebo realtime factor

talhabw, 2024-12-07:
gazebo real time factor starts at ~1.00 when the simulation is started. after some time it slows down to the level of ~0.05. possible causes may be:
- unnecessarily big sea world
- something related to our usage of gazebo
- some problem in gazebo 11 unrelated to us

we need to determine the reason and fix it if possible.

senceryazici, 2024-12-07:
Lets try the pool environment and try to eliminate the first cause

senceryazici, 2024-12-07:
Also what about the cpu usage? Does it start low and go up when rtf drops? Or is it always high?

ozanhakantunca, 2024-12-10:
Could you add your findings as comments under the card?  @eminmeydanoglu @optimistprime21 

Did the issue get resolved after transitioning to the pool environment?  
What is the average real-time factor of the data we are receiving?  
As Sencer Yazici mentioned, how is the CPU usage?  
What conclusions have you reached based on your research?  

Adding these outputs to the card would help us make better progress.

eminmeydanoglu, 2024-12-11:
### Some observations
1. In seabed, rtf is pretty much always low, even at the start.
2. In pool world, rtf is pretty high and stable. 
3. Couldn't observe the rtf going down with time, at least with the pool world. The cpu usage is around 150 (I don't know what the number represents, check ss) and doesn't seem to go up with time. 

[image]

closed 2024-12-17 by talhabw