itu-auv/auv-software #128: bug: mission_start_link is almost identical to odom

eminmeydanoglu, 2025-02-04:
### Exloring around
Lets explore initialize.py, the first state in our SMACH:

### 1. Resetting the Odometry Pose
```py
            smach.StateMachine.add(
                "SET_START_FRAME",
                SetStartFrameState(frame_name="mission_start_link"),
                transitions={
                    "succeeded": "SET_ALIGN_CONTROLLER_TARGET",
                    "preempted": "preempted",
                    "aborted": "aborted",
                },
            )
```
The instance this state runs, odom and taluy_base_link becomes almost identical (with a pi/2 angle offset). and then over time, localization reintroduces the expected offset, gradually shifting the perceived base position. How this looks in 3D will be introduced lately in the video below.

### 2. Creating the `mission_start_link` frame
```py
            smach.StateMachine.add(
                "SET_START_FRAME",
                SetStartFrameState(frame_name="mission_start_link"),
                transitions={
                    "succeeded": "SET_ALIGN_CONTROLLER_TARGET",
                    "preempted": "preempted",
                    "aborted": "aborted",
                },
            )
```
`SetStartFrameState` essentially creates the frame `mission_start_link`, with a pose identical to `taluy/base_link` (since pose transformation is omitted) , and with the identity transformation applied
```py
transform_request.transform.transform.rotation.w = 1.0
```
Can you guess what happens when these two states are combined one after the other? Yes, the mission_start_link is basically the odom frame, with a pi/2 transformation in yaw applied. 

[image]

https://github.com/user-attachments/assets/2a4edbd6-24a4-481e-9e72-ae1317f3769c

This is surely not desired since odom does not represent the taluy frame, especially right after it was resetted. 
And the third state is;
```py
            # TODO add enable controller
            smach.StateMachine.add(
                "SET_ALIGN_CONTROLLER_TARGET",
                SetAlignControllerTargetState(
                    source_frame="taluy/base_link", target_frame="mission_start_link"
                ),
                transitions={
                    "succeeded": "succeeded",
                    "preempted": "preempted",
                    "aborted": "aborted",
                },
            )
```
which basically says the robot to go to odom (whilst rotating 90 degrees). This is why we see the robot trying to to odom in the video. (it can't, becuase of the set_depth state in smach we will introduce shortly)

So this was the last state in initialize.py

After that, we go on with the gate smach. First state of gate smach is,
```py
            smach.StateMachine.add(
                "SET_GATE_DEPTH",
                SetDepthState(depth=gate_depth, sleep_duration=3.0),
                transitions={
                    "succeeded": "SET_ALIGN_CONTROLLER_TARGET",
                    "preempted": "preempted",
                    "aborted": "aborted",
                },
            )
```
which now tells the robot to go to gate depth (and the align frame request is still active, trying to move the robot to 0,0,0,0,0,1.) Since the active alignment request causes the taluy/enable topic to be set to True, which is how depth control is also able to work. 
After the set depth request is sent, it sleeps for 3.0. So this is the case for 3 seconds. And after this, we request alignment with gate_target, and finally be freed from mission_start_link.

### Hmm.
We surely don't want to go to odom. So this definitely a problem. It made itself apparent whilst I was working on applying the follow path action server to gate.py. I unknowingly made the time between the requests of aligning to mission_start_link and dynamic_target longer (by first planning the route and only then setting the new alignment request), which resulted with the robot trying to go to odom and rotating crazy! I actually noticed that the robot was not successfuly aligning with gate_depth (which is again because of this bug), so decided to make the sleep_duration 5.0 seconds, which made things even worse since this also made the time between requests longer :D 
 
### solution?
Having the taluy/base_link instance at the smach launch is cool, but I cant see any good reason to align with mission_start_link (even it is placed correctly), since we are then immediately requesting to go to gate depth (so it is still a problem if mission_start_link was being placed correctly). Let's remove the alignment request with misssion_start_link entirely.

But not sending the alignment request means that taluy/enable won't be set True. Then, let's publish to enable topic whilst the state is running.

senceryazici, 2025-02-17:
If so, that is a good find! But apart from attempting a solution, I'm also curious why this didn't cause problems during the competition? or after? or did it? Can you take a look at that as well? @eminmeydanoglu

senceryazici, 2025-02-17:
As part of the solution, the odom should be the aligned with the pose when the odometry starts, yes. Our IMU provides already filtered angles and we wanted to make use of that. But that is global orientation. You cannot directly feed this information into the EKF. There should be an option called `relative` or something like that in the ekf.yaml. this will help EKF to offset the initial orientation, thus aligning both frames! 

Here is the catch, if the `is relative` is set true, it will not only perform this offset for the angles, but for all the components of the IMU? This can be a problem. But ideally,

`map` -> `odom` -> `base_link`

This should be the chain. the odom and base must align at the odometry start, where the `map`->`odom` transform is for aligning the map frame to the global orientation. So if no drifts were to happen (ideal world) the initial orientation reading of the IMU (init_rx, init_ry, init_rz) should be used to determine the map -> odom transformation. With proper configuration this could be solved within the EKF.yaml since we disabled the `map` frame for avoiding above problem.

eminmeydanoglu, 2025-02-17:
> However, the start link be placed immediately without waiting for that convergence?

Yes--this is a why to the problem. 
> I'm also curious why this didn't cause problems during the competition?

Maybe because there is also the set depth request to gate (again immediately after) meaning the alignment request to mission_start_link it is not the only control request in play. I don't have the competition bags with me now, but when I inspected some older test bags (from months ago), I did witness it happening (mission_start_link being at odom level).

Btw, even if mission_start_link is in the right place (where taluy really is), it doesn't change the fact that there are two control requests in play (then we want to both go to the old taluy instance and to the gate depth). So it *must* cause unexpected behaviour no matter what. So, while the misplacement of mission_start_link is a reason, it’s not the reason. It’s just one piece of the problem.

eminmeydanoglu, 2025-02-19:
Hey, also because we have this new line in align frame controller!
```py
twist.linear.z = self.constrain(trans[2] * kp, self.max_linear_velocity)
````
Funny how this popped into my head the moment I woke up from a late afternoon nap? :D

closed 2025-03-05 by eminmeydanoglu