itu-auv/auv-software #131: Robot is Not Buoyant, Causing Z-Axis PD Controller to Get Stuck

eminmeydanoglu, 2025-02-05:
taluy lacks buoyancy and tends to sink. This results in a situation where the net downward force (gravity - buoyancy) balances the PD upward force, causing the robot to get stuck below the target depth. The PD controller is unable to compensate effectively because the net force on the robot is zero at this incorrect depth.

#### 2 solutions are,

0. Fix taluy's buoyancy :)

1. Enable Integral Factor:

   The integral term will accumulate error over time, increasing the control output and hopefully pushing the robot out of the stuck position.

2. Calculate F = gravity - buoyancy force and constantly apply it as positive wrench:

   If we can calculate the missing force, we can add it up as a constant to the wrench vector. However, I don't think we will be able to dynamically measure the F force and will use a static value. So we will assume that buoyancy never changes. But it will be changing (decreasing) over time, for example because of the sponge (filling with water). 

I believe the best approach is a combination of both. The constant force option will resolve most of the issue, while the integral factor will perfect the alignment. 

Credits to @ozanhakantunca for pointing out the issue :)

eminmeydanoglu, 2025-03-06:
We encountered unexpected behaviour with I controller. Check #149

ozanhakantunca, 2025-04-29:
Check this: https://github.com/itu-auv/auv-software/issues/149#issuecomment-2838294549