itu-auv/auv-software #158: "red_buoy.py" should be refactored

ozanhakantunca, 2025-03-07:


eminmeydanoglu, 2025-03-09:
Is this a backlog task? We won't need it for this year's competitions, so we can leave it for later.

ozanhakantunca, 2025-03-09:
We still have active tasks related to performing rotational motion around an object. The object we rotate around may change depending on the detected target, and the node's name may vary, but we will still need these functions. Currently, I believe the functions defined in red_buoy.py are not structured optimally and are placed inappropriately. If we refine this system and make it more flexible, we can adjust ourselves to rotate around any detected object. As a result, this process also implies that the file needs to be refactored.

closed 2025-05-02 by ozanhakantunca